PEEC-Nature-Youth-Group/pumaguard #8: Research how much power solar panel modules can deliver and how much they cost

nicolasbock, 2024-09-19:
We need to supply the Raspberry Pi's with power. We might be able to do that using solar panels but we need to understand how much power we could produce with them.

celiapesiri, 2024-10-01:
A general rule of thumb is that solar panel modules provide **1,000 watts (1 Kw) of sunlight for every square meter of panel.**  Depending on how much space on our design we have for solar panels, this is something we can consider. We also need to figure out how much energy is required for Raspberry Pi to run our images and if that would even be possible with the amount of memory and software we have. I will do some more research on this and add it into a second comment.

nicolasbock, 2024-10-02:
Thanks @celiapesiri . It might help to also see what solar cell manufacturers claim their solar cells can deliver. The typical efficiency of solar cells is probably no more than 20% which would reduce the number you found to around 200 W/m^2.

tatep0, 2024-10-24:
1000 Watts per square meter is the total amount of energy coming from the sun but most solar panels are at around 20% efficiency. The value of a solar panel is usually measured by dollars per Watt it produces. If size isn't a problem, then we can go with a lower efficiency solar panel that is cheaper (lower dollars per Watt value).

tatep0, 2024-10-24:
I made a mistake in the last post. Watts is energy rate and Watt hours (Wh) is energy. Secondly, the raspberry pi will require 0.5 amps at a time. If we assume the battery won't be able to charge for 16 hours max, we need an 8 Ah battery. The battery will also need to output 2.5 Watts so it needs to be able to store 40 Wh but we need a buffer so 50 Wh is better. Below are the exact formulas.

Assuming 16 hours between a full charge:
Q (charge stored by battery, amp hours) = I (current, amps) * T (time, hours) = 0.5 amps * 16 hours
E (energy stored by battery, watt hours) = P (power, watts) * T (time, hours) = 2.5 Watts * 16 hours