PEEC-Nature-Youth-Group/pumaguard #9: How high is the power consumption of a Raspberry Pi Zero W?

nicolasbock, 2024-09-19:
Find out how much power a Raspberry Pi Zero W consumes when idle and when under load. This will help us understand better how much total power we will need.

adisbock, 2024-09-26:
Power consumption of the Rasberry Pi Zero W

100 mA: No peripherals connected, WiFi And Bluetooth off
160 mA: Bluetooth turned on
170 mA: WiFi turned on
230 mA: Camera connected and capturing

nicolasbock, 2024-09-26:
Thanks @adisbock @seblkoglin !

Amperes is the current the units draw, not the power. If we know the voltage we can calculate power with

```
P = V * I
```

Could you find out what the voltage is?

Then, open a pull request in which you add this information to [the parts list](https://github.com/PEEC-Nature-Youth-Group/PumaGuard/blob/main/docs/source/Parts.rst).

nicolasbock, 2024-10-21:
Can one of you create a pull request to add these numbers to the webpage?