PEEC-Nature-Youth-Group/pumaguard #80: Put together parts list for building our own camera

nicolasbock, 2024-10-03:
Create a parts list for Raspberry Pi based trailcam.

seblkoglin, 2024-10-03:
[Arducam Day-Night Vision for Raspberry Pi Camera, Automatic IR-Cut Switching All-Day Image All-Model Support, IR LED for Low Light and Night Vision, M12 Lens Interchangeable, OV5647 5MP 1080Phttps://      
www.amazon.com/dp/B07X1VGQSL?ref=ppx_yo2ov_dt_b_fed_asin_title

 CQRobot 10.525GHz Doppler Effect Microwave Motion Sensor Compatible with Raspberry Pi and Arduino Board. for Industrial, Measurement, Automatic Doors, Sensor Lights, Reversing Radar, etc.
https://www.amazon.com/dp/B089NKGWQQ?ref=ppx_yo2ov_dt_b_fed_asin_title

 
[Gikfun 2" 4Ohm 3W Full Range Audio Speaker Stereo Woofer Loudspeaker for Arduino (Pack of 2pcs) EK1725]
https://www.amazon.com/dp/B01CHYIU26?ref=ppx_yo2ov_dt_b_fed_asin_title

 
 [Teyleten Robot AMG8833 8 * 8 IR Infrared Temperature Thermal Imager Array Camera Imaging Sensors Module I2C 3-5V for Raspberry Pi Arduino]
https://www.amazon.com/dp/B0C7KXHXTP?ref=ppx_yo2ov_dt_b_fed_asin_title

nicolasbock, 2024-10-04:
I'll reopen this issue so we can copy it into the webpage. @seblkoglin do you want to do that?

nicolasbock, 2024-10-04:
And thanks for all of the links! Makes this much easier :smile:

closed 2024-10-17 by nicolasbock